MrYsLab/Telemetrix4Arduino #1: Analog scan interval does not play nicely with differential

mklomp, 2020-11-10:
When setting the scanning interval to 20, and the analog differential to 0, I'd expect to get updates every 20ms (50Hz). I only get the, at around 24Hz. the problem is that the code is still checking whether the new value changed in:

https://github.com/MrYsLab/Telemetrix4Arduino/blob/275f58b98e336855deb677ae5f7406e5a253b469/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L881

While will also be handled when setting the differential to 1:

https://github.com/MrYsLab/Telemetrix4Arduino/blob/275f58b98e336855deb677ae5f7406e5a253b469/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L885

My expected behaviour is when I remove the first check completely.

MrYsLab, 2020-11-15:
Fixed with version 1.3

mklomp, 2020-11-17:
Thanks for fixing this. I see in your current implementation you decided to check if the differential is set. The fix is then to also send a message when differential is note set (eg. 0). Althoug this indeed fixes the problem, I was thinking about another solution.

Since the introduction of the differential there seems to be no need for the logic to check if the value is the same as the last value. At first this last_value logic was introduced to minimize the number of messages. But the differential will do this already for you, so there is no need to use this anymore. When differential > 0, duplicates will not be sent anyway. So, I think changing;

https://github.com/MrYsLab/Telemetrix4Arduino/blob/f4370d5be3befdf58765e369cf6ddc49d77d9c01/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L891-L921

Could just be replaced by: 

https://github.com/MrYsLab/Telemetrix4Arduino/blob/f4370d5be3befdf58765e369cf6ddc49d77d9c01/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L896-L908

As far as I understand the logic then stays the same, and the complexity of the code is reduced. Or am I missing something?

MrYsLab, 2020-11-20:
Release 1.4 solves this issue.

closed 2020-11-20 by MrYsLab