MrYsLab/Telemetrix4Arduino #3: Only able to run script with US sensor 6 times

mklomp, 2020-12-09:
I can run the example [script](https://github.com/MrYsLab/telemetrix-aio/blob/master/examples/hc-sr04_distance_sensor.py) perfectly fine. After stopping it with CTRL-C (invoking a board.shutdown()) I can restart it again, and it still works fine. I can only do this 6 times, since each call of the script will add a US filling up to MAX_SONARS. This of course is due to the fact hat the MCU does not reset and keeps its state.

As far as I can see I can think of three possible ways to solve this:
- Use a 'soft reset' when initializing (or on shutdown) resetting all pins to their defaults, or at least emptying MAX_SONARS etc.
- Detect when a new sensor has the same config (pins) as a config already in use.
- Use a 'hard reset' really resetting the whole arduino on the shutdown.

This probably also holds for MAX_DHT and MAX_SERVO.

MrYsLab, 2020-12-09:
Thanks for the report. Probably, the simplest solution is to send a message from the client to the server when the client first starts to tell the server to reinitialize its data structures. I will try this approach and if successful, will publish a new version for both client and server.

MrYsLab, 2020-12-11:
Resolved with release 1.9

closed 2020-12-11 by MrYsLab