MrYsLab/Telemetrix4Arduino #4: Multiple servos not working correctly

mklomp, 2021-02-19:
When having multiple servos attached, I get the behavior that always the first one is also moving. This line seems to be the cause:

https://github.com/MrYsLab/Telemetrix4Arduino/blob/15a07fa747f7b74b5bcd28fd9373445a6fe48626/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L510

MrYsLab, 2021-02-19:
@mklomp Thanks. I will remove the offending line and test. If all works well, I will publish a new release. I hope you and the family are well.
FYI - I am currently working on a Telemetrix client and server for the Raspberry Pi Pico board. I just got the basics working (flashing the board LED). I am using the Pico C SDK.

MrYsLab, 2021-02-19:
This is now fixed with release 1.10. It may take a day or so for the Arduino library manager to pick it up. Thanks for the bug report.

closed 2021-02-20 by MrYsLab