MrYsLab/Telemetrix4Arduino #15: Sampling rate not correct at start

mklomp, 2022-05-12:
When using a single analog input icm a sampling rate, the sampling rate is not correct at start. After some time it will move to the set sampling rate. When using the analog.py example from Telemetrix-AIO the callback can be changed to print the frequency of the sensor every second. When setting the sampling rate to e.g. 200 it will not print 5 (Hz) from the start, but the max frequency for some time. When setting the sampling rate lower, it takes even longer for it to get to the right frequency.

As far as I could see (and test) this line:

https://github.com/MrYsLab/Telemetrix4Arduino/blob/68ce3ab7d2e50f17cc0f30c0ac90a507633975fd/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L1780

should be

`previous_millis = current_millis;`

possible icw initializing previous_millis to 0.

Please note that all sensors seem to have this kind of mechanism.

MrYsLab, 2022-05-12:
Thanks. I will investigate.

MrYsLab, 2022-05-12:
This issue is resolved by release 5.1.1. Thanks for the report.

closed 2022-05-23 by MrYsLab