MrYsLab/Telemetrix4Arduino #17: Command corruption when sending many commands in a row

Xevel, 2022-05-23:
Hello,

I'm using the version in the Arduino Library Manager (5.1.1) on an Arduino Mega 2560 (either original or third party 2560 Pro Embedded, which uses a CH340 USB to UART chip).
The Python code driving everything uses your telemetrix code currently in the master branch though I'm not sure about the version. 
I'm running the code on a RPi4 B+ running the standard Raspbian, with X server, and only PyCharm running.

My python code initializes 16 pins immediately after instantiating Telemetrix, and sets each as a digital output, with an initial state.
Later (delay of 1s) I write one of these pins to change its state, and the problem is that this action changes both the state of the expected pin AND the state of ANOTHER pin (always the same other pin, regardless of which pin in intended to change...).
I did not investigate too much but it's been 100% reproductible so far, on 2 2560 Embedded Pro and one official Mega 2560.

In the attached example, setting pin 39 to HIGH also turns pin 45 to HIGH, but after that when pin 39 is toggled, pin 45 stays HIGH.
If I add a small (l.38) the problem seems to disappear. Same if I remove the commands that set the initial pin state.
I suspect the number of commands is more important in this problem than the actual commands...

```
  from telemetrix.telemetrix import Telemetrix
  import time
  
  class PinDef:
      def __init__(self, pin, init_state_or_pullup=True):
          self.pin = pin
          self.init_state = init_state_or_pullup  # init state when in output, pullup if in digital input
  
  digital_output_pins = {
      "RELAY_A": PinDef(38, False),
      "RELAY_B": PinDef(39, False),
      "RELAY_C": PinDef(40, False),
      "RELAY_D": PinDef(41, False),
      "RELAY_E": PinDef(42, False),
      "RELAY_F": PinDef(43, False),
      "RELAY_G": PinDef(44, False),
      "RELAY_H": PinDef(45, False),
      "RELAY_I": PinDef(46, False),
      "RELAY_J": PinDef(47, False),
      "RELAY_K": PinDef(48, False),
      "RELAY_L": PinDef(49, False),
      "RELAY_M": PinDef(28, False),
      "RELAY_N": PinDef(29, False),
      "RELAY_O": PinDef(30, False),
      "RELAY_P": PinDef(31, False),
  }
  
  board = None
  
  def telemetrix_init():
      global board
      board = Telemetrix()
  
      for name, pin_def in list(digital_output_pins.items()):
          print("Setup Digital pin", name, pin_def.pin, pin_def.init_state)
          board.set_pin_mode_digital_output(pin_def.pin)
          board.digital_write(pin_def.pin, pin_def.init_state)
          # time.sleep(0.1)
  
  def relay_set(pin, state):
      if state:
          print('Turn ON relay', pin)
          board.digital_write(pin, 0)
      else:
          print('Turn OFF relay', pin)
          board.digital_write(pin, 1)
  
  if __name__ == "__main__":
      telemetrix_init()
  
      time.sleep(1)
  
      while True:
          relay_set(39, True)  # here on the first execution, both pin 39 and 45 are set HIGH.
          time.sleep(0.5)
          relay_set(39, False)  # only pin 39 is affected, as expected
          time.sleep(1)
```

Best,
Nicolas

MrYsLab, 2022-05-23:
Hi Nicolas,
     I don't see anything obvious that would cause this. If you comment out
```
          board.digital_write(pin_def.pin, pin_def.init_state)
```
in the _for loop,_ are all pins set to LOW?

If they are, and you then set pin 39 high in relay_set, does pin 45 also go high by itself?

Are you using relays with coils? If so, I wonder if an inductive kick could cause issues when the relay is activated?

MrYsLab, 2022-05-23:
Re-reading your description and giving this some additional thought, it may be that the serial receive buffer in the Mega is being overwritten somehow due to the rate of commands being sent. Adding a delay to **_fix_** the problem suggests that the data is being corrupted. However, this makes little sense since each command consists of packet length, command value, and possible payload. If the packets are corrupted, I don't see how the sketch does not crash. 
If your application can handle adding a short delay and if this solves the issue, I suggest using this approach.

MrYsLab, 2022-05-23:
According to [this posting](https://forum.4dsystems.com.au/node/51986), the serial buffer sizes can be set with a #define at the top of the sketch.

```
#define SERIAL_RX_BUFFER_SIZE 128
#define SERIAL_TX_BUFFER_SIZE 128
```
You might give that a try and see if it solves your problem.

Xevel, 2022-05-24:
In the end, for my application, I implemented an ACK for all commands. It's a very simple fix that gives me at least some synchronization and some health info on the communication.
Given the nature of the project, it's not a change that can be easily propagated to all the related project... but if you're interested I can submit a PR.

MrYsLab, 2022-05-24:
Thanks for letting me know, and thanks for the PR offer. I will leave the communication as is because there is very little room left when compiling for the Uno, and I wish to leave space for any bug fixes that might come along. Thanks again.

Xevel, 2022-05-24:
On the Arduino side, the change is to add, at the very end of get_next_command(), something like
```
  Serial.write(0x02);
  Serial.write(ACK);
  Serial.write(command);
```
So it might be acceptable ?

On the Python side, you have many ways to implement it. Sending a command resets an instance variable, sends the command then waits for that instance variable to change back to the sent command value (or timeout).
The _reporter() thread simply fills that instance variable with the argument of the ACK packet, which has to match the previous command sent.
You get synchronization and basic validation that the command has been at least heard by the server.

Cheers.

MrYsLab, 2022-05-24:
Hi Nicolas,
    I agree that the coding is somewhat trivial. I am more concerned about the implication on the design that implementing this feature raises.

First, for users with existing applications, this will affect their timing. It may not matter in some cases, but I would think that it may be an issue for some.

So does the ACK feature need to be provided with the choice to turn it on and off globally? Perhaps, a parameter could be provided that activates the ACK selectively with each command call.

Then either the user or the Python library code would need to keep track of which commands will be returning the ACK.

In any case, what happens if an ACK is never returned? Is the command automatically resent?

All these questions are answerable, but I still think adding a short delay after sending each command is simpler. 

Alan

Xevel, 2022-05-24:
All good reasons.

Thank you for your work :)