MrYsLab/Telemetrix4Arduino #18: Digital Pin 13 does not work as digital input on Arduino Mega 2560

Xevel, 2022-06-03:
Hi,

I'm using an arduino Mega 2560 Pro embedded, and I came accross a very peculiar behaviour.

I set up multiple pins as digital_input_pullup (13, 53, 57(=A3)), and one of them behaves very differently from the others: it goes to 1.8V (instead of the 5V the pullup should pull to) and does not generate any digital report when shorted to GND ( though that seems reasonable in any case since it is always always below V_IL).

The other pins behave as expected.

Another oddity, when the arduino is reset by the telemetrix lib connecting to it, when all the other pins (digital input) stay LOW until the command to use the pullup is received, that pin 13 goes HIGH and LOW twice for about 500ms (from memory, I can provide go back and make measurements if useful).

FYI I have added an ACK mechanism (discussed already in a telemetrix repo issue), that ensures that no command is missed or corrupted silently. 
Tested on multiple ardunio units running the code in the repo at this date.

MrYsLab, 2022-06-03:
If you write a sketch that directly sets the pins to digital inputs, is the behavior the same as with Telemetrix? If not, which of the pins are behaving differently? Also, if you disconnect the hardware attached to those pins, do the pins act as expected?

Telemetrix does not actually reset the hardware (it can't). With pin 13 going high after a pull-up command is received, which pin is pulled up? No need to do a measurement just yet.

MrYsLab, 2022-06-03:
I found this comment [on this link.](https://forum.arduino.cc/t/can-i-use-pin-13-as-an-input/348096/8) Could that explain what you are seeing?

On the official Uno, the pin13 LED is buffered with an opamp, so you can use pin13 freely. On many boards though, the LED is just connected directly to the pin, and the load from the LED prevents it from being a useful input.

The symptom of this would be with it set as INPUT_PULLUP, the LED would light very dimly, and digitalRead() would always look low.

Xevel, 2022-06-03:
I do feel silly, that's a newbie mistake on my part to forget about that LED... and to fail to find that answer by myself. It's been too long I haven't used an Arduino.

Indeed that's the reason. Thanks a lot.

MrYsLab, 2022-06-03:
No reason to feel silly. After turning on a servo motor on an Uno, I once spent three weeks chasing down a PWM issue. There is a note in the Servo library documentation that I missed that reads:

_On boards other than the Mega, use of the library disables analogWrite() (PWM) functionality on pins 9 and 10, whether or not there is a Servo on those pins._ 

Talk about feeling silly ;-).

closed 2022-06-03 by MrYsLab