MrYsLab/Telemetrix4Arduino #20: Arduino Mega2560 - RX pin lose bytes

zxia545, 2022-11-23:
I have config 15 Analog pin to Analog input pin and then I found sometime Arduino did not received correct message that I sent from Python. 
It seems like a problem with continue reporting back analog pins and digital pins from TX back to python. The RX sometime will lose few bytes.

So, I used oscilloscope to monitor RX pin from Arduino, and sending [0x91, 0x40, 0x01] and [0x91, 0x00, 0x01] each 0.5 second. But sometimes it will drop the last or last two bytes.
[image]

This problem can be solved by changing the baud rate from 115200 to 9600, but then for those pins be config as analog input become a problem since the baud rate is dropping. Also, another way to hide this issue is only setting one or two pins to be analog input pin.

MrYsLab, 2022-11-24:
Thanks for reporting tthis. I don't have the resources to run that number of analog inputs, but this suggestion may solve the issue. You can change the analog input scan rate by calling [set_analog_scan_interval](https://htmlpreview.github.io/?https://github.com/MrYsLab/telemetrix/blob/master/html/telemetrix/index.html#telemetrix.telemetrix.Telemetrix.set_analog_scan_interval)  The default scan rate is 19 ms.

Please let me know if this solves the issue.

zxia545, 2022-11-24:
Thanks a lot, but I am currently using the **pymata4** with **firmata-express** :(. 
So can I call set_sampling_interval in pymata4 to achieve the same result? And what if disable all the analog input pins polling that Arduino only sends analogRead value when receiving an inquiry for an analog pin (a command send from pymata4)?

Also wondering why you decided to use the polling mechanism for reading analog input and digital input. Is it because you want to increase the speed of the python reading value of analog and digital pins?