MrYsLab/Telemetrix4Arduino #23: i2c_write to MCP4725

phthalimid, 2023-01-03:
Hi Alan,

I really love Telemetrix. Great work! However, I've got one major issue: I didn't manage to write to a DAC chip (MCP4725) via i2c_write(). Whatever args I give, I get always the same error message:

"TypeError: 'bytes' object cannot be interpreted as an integer"

Here is my code fragment:

mcp4725 = b'\x60'
buffer = [0, 0]
board.i2c_write(mcp4725, buffer)

As 'buffer', I tried pretty much everything that came to my mind. Some examples:

buffer = [0]
buffer = [b'\x00', b'\x00']
buffer = bytearray(b'\x00\x00')
... and many others. They all resulted in the same error message.

I hope you can help.

phthalimid, 2023-01-03:
I forgot to post the whole error message:

TypeError                                 Traceback (most recent call last)
<ipython-input-40-33476f791aeb> in <module>
      4 buffer = [0]
      5 
----> 6 board.i2c_write(mcp4725, buffer)

~/opt/anaconda3/lib/python3.8/site-packages/telemetrix/telemetrix.py in i2c_write(self, address, args, i2c_port)
    700             command.append(item)
    701 
--> 702         self._send_command(command)
    703 
    704     def loop_back(self, start_character, callback=None):

~/opt/anaconda3/lib/python3.8/site-packages/telemetrix/telemetrix.py in _send_command(self, command)
   2281         # the length of the list is added at the head
   2282         command.insert(0, len(command))
-> 2283         send_message = bytes(command)
   2284 
   2285         if self.serial_port:

TypeError: 'bytes' object cannot be interpreted as an integer

MrYsLab, 2023-01-03:
@phthalimid Thanks for the question. My choice of words for the comments for the parameters of the i2c_write method could be more precise, and I will explain things more precisely in a future release. 

All parameter values are expressed as integers limited to 0-255. That is what I meant by bytes and not the byte type.

So for your example, the call to i2c_write would look like the following:
```
mcp4725 = 0x60
buffer = [0, 0]
board.i2c_write(mcp4725, buffer)

```

Sorry for the confusion. Also, I am moving this issue to the telemetrix repository. 

Please let me know if this solves your problem.