MrYsLab/Telemetrix4Arduino #26: Does not compile with Arduino Uno R4

speccy88, 2023-09-01:


MrYsLab, 2023-09-01:
You are correct. There is separate repository for the [Arduino R4 boards](https://github.com/MrYsLab/telemetrix-uno-r4). A user manual may be found [here](https://mryslab.github.io/telemetrix-uno-r4/).

Please let me know if you have any further questions.

speccy88, 2023-09-01:
Thanks,

I have a flashed Tasmota on my uno r4 esp32 and can communicate with the renesas chip using uart. I have soldered the usb bridge under the board and installed the minima bootloader so I can upload arduino sketch directly from usb.

I would like to use the berry scripting feature on tasmota to have a telemetrix berry client that communicates with the arduino. I would start by basic digital pin control. Any pointers for how to start?

Fred

MrYsLab, 2023-09-02:
Telemetrix uses the standard USB bridge configuration of the R4 WIFI, so it is "unaware" that the ESP even exists. 
In your configuration, if you can run any of the Telemetrix examples, such as [loopback for the serial transport ](https://github.com/MrYsLab/telemetrix-uno-r4/blob/master/telemetrix_uno_r4/r4_wifi_examples/threaded/usbSerial/wst_loop_back.py)or [loopback for the WIFI transport](https://github.com/MrYsLab/telemetrix-uno-r4/blob/master/telemetrix_uno_r4/r4_wifi_examples/threaded/WIFI/wwt_loop_back.py), then you need need to extend both the Telemetrix client and server to send and receive Berry messages. You would then need to relay the Berry content to and from the ESP. If you need help extending the Telemetrix messaging protocol, I would be happy to answer your questions, but I will be of limited use with the Tasmota and Berry side of things.
Since this is not a Telemetrix issue, I will be closing the issue, but if you have any questions, feel free to post them here, and I will see them.

closed 2023-09-02 by MrYsLab