MrYsLab/Telemetrix4Arduino #27: Stepper reverses direction after 2**15 steps

dirtbirb, 2023-09-28:
I'm using a stepper on a lead screw with very fine threads. It looks like if I tell it to go beyond 2**15 steps, it turns around and goes the opposite direction. I assume this is a short int thing, interpreting larger numbers as if the sign got flipped. If that's accurate, any reason not to change that to a longer data type, or is there some best practice I should prefer instead?

I'm running this on an Arduino Nano via a PC app in Python. Happy to provide any other relevant info.

Thanks!

MrYsLab, 2023-09-28:
Which command is exhibiting this issue?
The following commands all use 32 bits (at least they should be):

1. stepper_move_to
2. stepper_move
3. stepper_set_current_position

dirtbirb, 2023-09-28:
stepper_move_to, can try others. If those are all 32 bits, then it's likely
this is occurring somewhere else, like in kivy (GUI framework) instead.
Will update when I have more info (probably next week).

Thanks for the response! Mostly it's just good to confirm that those are
indeed 32 bits (probably).

MrYsLab, 2023-10-08:
I am closing this issue. If you are still having problems, please post a simple code example so I may test it, and I will reopen the issue.

closed 2023-10-08 by MrYsLab