MrYsLab/Telemetrix4Arduino #28: Limited to 3 steppers on 5.3.2+

StephenB1289, 2023-10-12:
First off, thank you for creating and maintaining this excellent library!

It seems in version 5.3.2, the change to allow up to 8 steppers has somehow actually limited the number of steppers that can successfully be created to 3. Creating a 4th stepper doesn't generate an exception, but does seem to prevent the Arduino from any further operations (including operating steppers 0 – 3). Reverting to 5.3.1 allows a 4th stepper to work just fine.

I took a look at the commits for 5.3.2, and while they seem basic, was unable to determine the root cause for this... Both telemetrix and telemetrix-aio are affected, so I believe this is caused by some issue in the Arduino sketch. Unfortunately, I'm not familiar enough with C++ to dive into why changing the MAX_NUMBER_OF_STEPPERS constant is having this effect.

Please let me know if you need more information or are unable to replicate this. Thank you for your hard work!

MrYsLab, 2023-10-13:
You are correct. I can reproduce the issue, which is definitely somewhere in the Arduino sketch. It may take me a few days to figure out what is happening. I will post here when I do. Thanks for reporting this.

MrYsLab, 2023-10-13:
@StephenB1289 I found the issue. In testing on an Arduino UNO, insufficient heap space exists to allow for the dynamic allocation of more than 4 AccelStepper objects. Trying on a Leonardo and Mega2560, there is enough space for allocating eight objects, so this problem does not exist on those boards.

What Arduino model are you using?

I will update the documentation to reflect this limitation of the UNO.

MrYsLab, 2023-10-13:
@StephenB1289 One more point - for an Arduino UNO, the maximum number of steppers is 3.

StephenB1289, 2023-10-14:
@MrYsLab Wow, thank you for looking into this so quickly! I suppose I should've been able to figure that out myself, now that I look at the source again... Indeed, I did only test this using UNO boards.

I'm not sure that I follow your second point, though – do you mean the max number of steppers for an UNO running Telemetrix4Arduino is **_4_**? If I'm running version 5.3.1 I have no issues driving 4 steppers with an UNO.

In any case, thank you again for taking a look! This library is incredibly helpful.

MrYsLab, 2023-10-14:
The reduction to 3 is because the latest version, 5.3.3, added some code and made less memory available for stepper instances.

You may be able to reclaim some memory space if you are using only some of Telemetrix's features. If you comment out any of the [component defines](https://github.com/MrYsLab/Telemetrix4Arduino/blob/6883fcd978cfb5a31882a6f0b335f099df1e8d1a/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L41) the code for that feature will not be compiled. I don't know how much can be reclaimed, but it may help. 

I am closing this issue, but you may comment here, and I will see it.

closed 2023-10-14 by MrYsLab

StephenB1289, 2023-10-17:
I apologize for the delay in responding, but I've just had the chance to test this out again. I'm surprised the STEPPER_SET_SPEED patch was enough to cause the memory issue to recur; apparently the full sketch really pushes the limits of the UNO's memory!

Disabling some unneeded component defines has allowed me to free up plenty of space on an UNO to drive 4 steppers. All is functioning properly on (a slightly modified) 5.3.3! Thank you again for your assistance.